scikit-rf/scikit-rf #1249: 5 Port MDIF File fails to import to NetworkSet

jrmagers, 2025-02-08:
**Describe the bug**

A 5-port MDIF file created from the s-parameters of the SP4T switch [ADRF5054](https://www.analog.com/en/products/adrf5045.html). This part is confirmed to correctly import into AWR Microwave Office.

**To Reproduce**
```Python
import skrf

filename = "ADRF5054.mdf"
adrf5054 = skrf.NetworkSet.from_mdif(filename)

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
File Untitled-1:6
      [3](untitled-1:3) import skrf
      [5](untitled-1:5) filename = "ADRF5054.mdf"
----> [6](untitled-1:6) adrf5054 = skrf.NetworkSet.from_mdif(filename)

File c:\venv\sandbox\Lib\site-packages\skrf\networkSet.py:348, in NetworkSet.from_mdif(cls, file)
    [329](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:329) """
    [330](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:330) Create a NetworkSet from a MDIF file.
    [331](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:331) 
   (...)
    [345](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:345) 
    [346](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:346) """
    [347](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:347) from .io import Mdif
--> [348](file:///C:/venv/sandbox/Lib/site-packages/skrf/networkSet.py:348) return Mdif(file).to_networkset()

File c:\venv\sandbox\Lib\site-packages\skrf\io\mdif.py:86, in Mdif.__init__(self, file)
     [84](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:84) self._comments = self._parse_comments(fid)
     [85](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:85) self._params = []
---> [86](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:86) self._networks = self._parse_mdif(fid)

File c:\venv\sandbox\Lib\site-packages\skrf\io\mdif.py:346, in Mdif._parse_mdif(self, fid)
    [340](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:340) # parse numerical data:
    [341](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:341) #
    [342](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:342) # begin XXX
    [343](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:343) # ....
    [344](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:344) # end
    [345](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:345) if line.lower().startswith('end'):
--> [346](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:346)     ntwk = self._parse_data(block_data)
    [347](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:347)     ntwk.params = params
    [348](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:348)     ntwks.append(ntwk)

File c:\venv\sandbox\Lib\site-packages\skrf\io\mdif.py:233, in Mdif._parse_data(self, block_data)
    [227](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:227)     # Data are described as
    [228](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:228)     #
    [229](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:229)     # v_kind1 v_kind2 ...
    [230](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:230)     # ...
    [231](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:231)     # ... v_kindN
    [232](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:232)     elif line.strip():  # ignore lines with only whitespace
--> [233](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:233)         data_lines.append(np.array(line.split(), dtype='float'))
    [235](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:235) # grouping the data_lines in a single line for each frequency
    [236](file:///C:/venv/sandbox/Lib/site-packages/skrf/io/mdif.py:236) nb_lines_per_freq = len(kinds)

ValueError: could not convert string to float: 'n15x'
```

**Expected behavior**

A `NetworkSet` object should be returned with parameters according to the the selected switch path (RF1, RF2, RF3, RF4).

**Data**

See attached MDIF (compressed as a zip).

**System**
 - scikit-rf version 1.5.0.

[ADRF5054.zip](https://github.com/user-attachments/files/18721822/ADRF5054.zip)

FranzForstmayr, 2025-02-11:
Has anyone access to some "official" specification of the mdif format? I cannot find it anywhere unfortunately.

However when looking at this document it looks like mdif does not support n-ports: (https://www.flowcad.de/datasheets/AWR-User-Guide.pdf)

[image]

To support n-ports we probably need to support the gmdif format in general.

FranzForstmayr, 2025-02-11:
You created this mdif file with skrf?
I think the issue is, that we write a wrong file rather than reading it wrong...

```

!-------------------------------------------------------------------------------
VAR name = "ADRF5054_RF1selected"

BEGIN ACDATA
%F n11x n11y n12x n12y n13x n13y n14x n14y 
n15x n15y 
n21x n21y n22x n22y n23x n23y n24x n24y 
n25x n25y 
n31x n31y n32x n32y n33x n33y n34x n34y 
n35x n35y 
n41x n41y n42x n42y n43x n43y n44x n44y 
n45x n45y 
n51x n51y n52x n52y n53x n53y n54x n54y 
n55x n55y 

! network name: ADRF5054_RF1selected
! Created Mon Apr 10 13:55:33 2023
!
! Created with skrf (http://scikit-rf.org).
# Hz S RI R 50.0 
!freq ReS11 ImS11 ReS12 ImS12 ReS13 ImS13
! ReS14 ImS14 ReS15 ImS15
! ReS21 ImS21 ReS22 ImS22 ReS23 ImS23
! ReS24 ImS24 ReS25 ImS25
! ReS31 ImS31 ReS32 ImS32 ReS33 ImS33
! ReS34 ImS34 ReS35 ImS35
! ReS41 ImS41 ReS42 ImS42 ReS43 ImS43
! ReS44 ImS44 ReS45 ImS45
! ReS51 ImS51 ReS52 ImS52 ReS53 ImS53
! ReS54 ImS54 ReS55 ImS55
!
50000000.0 0.05904910443269358 -0.012678435824380224 0.8903077781424098 -0.03364196589444086 4.798226024274454e-05 -7.362710809593013e-05 -1.8679530068304105e-05 6.148020376242954e-06
 -2.5166639320407287e-06 4.3317989789322456e-05
 0.8812052780655182 -0.031249734806021907 0.072038461366285 0.01809481079794019 9.458595908761723e-05 8.671762860388856e-05 -1.3356889669411835e-05 -1.1045068941911706e-05
 -6.763462686530363e-06 7.392413616581119e-06
 3.174597195859412e-05 -8.740641291808274e-05 9.20862601922844e-05 6.855401534909781e-05 -0.8952408426210837 0.006109430857573712 0.00011035091137076695 0.00011300341790782059
 -3.231065083060009e-05 1.3845025160532366e-05
```

After `BEGIN ACDATA` the parser expects float, but due to the high port number there's still the trace name string.

jrmagers, 2025-02-12:
> You created this mdif file with skrf? I think the issue is, that we write a wrong file rather than reading it wrong...

Yes, I created with scikit-rf.

This file does correctly import into AWR Microwave Office as a "Generalized N-Port MDIF" file:

[image]

I suppose that doesn't necessarily mean that the format is correct or conforming to the standard.  However, I have found the Microwave Office will reject incorrectly formatted Touchstone files.

FranzForstmayr, 2025-02-12:
So probably we can fix it by just removing the offending lines, at least to be compatible with AWR.

jrmagers, 2025-02-12:
The AWR Microwave Office user guide has pretty good documentation on the GMDIF file format. I have access so it, but I'm unsure if it is acceptable to share it.  I will try contacting Cadence support and ask permission to share it. (The documentation does include a "Share" button on it).

Unfortunately, I could not find any references to a specification for the GMDIF format in the document.

gurthang2, 2025-03-08:
I also ran into this issue with a GMDIF/MDF file generated from Microwave Office (I don't fully understand the difference to be honest). MWO distinguishes "new lines that aren't new lines" by starting them with a tab characters. I think that's where the issue is coming from in skrf. 

 I found a work around by appending lines that start with '\t' to the previous line. After you concateneate everything together the `rf.NetworkSet.from_mdif` function works as expected.

Before preprocessing:
```
! AWR Design Environment (17603) Fri Mar  7 16:50:01 2025
! Document = 'a_base_EM'
! nPorts: 3, nXvals: 6

VAR x1 = 20
VAR x2 = 30
BEGIN ACDATA
# GHz S DB R 50
% F N11X N11Y N12X N12Y N13X N13Y N21X N21Y 
	N22X N22Y N23X N23Y N31X N31Y N32X N32Y 
	N33X N33Y 
0.1 -9.5574863 179.87028 -3.5265759 -0.08911217 -3.524635 -0.042428161 -3.5265759 -0.08911217 
	-9.5573313 179.84258 -3.5245584 -0.05627355 -3.524635 -0.042428161 -3.5245584 -0.05627355 
	-9.5534535 179.93604 
1.1 -9.5640098 178.64805 -3.5296304 -0.95062039 -3.5257785 -0.45939033 -3.5296304 -0.95062039 
	-9.5651793 178.36214 -3.5264164 -0.60217335 -3.5257785 -0.45939033 -3.5264164 -0.60217335 
	-9.5582923 179.34614 
2.1 -9.5689254 177.45568 -3.5320908 -1.8007848 -3.5267782 -0.87172728 -3.5320908 -1.8007848 
	-9.5705455 176.91594 -3.5277627 -1.14117 -3.5267782 -0.87172728 -3.5277627 -1.14117 
	-9.5627602 178.77766 
3.1 -9.5730569 176.28011 -3.5347768 -2.6454479 -3.5278218 -1.2814322 -3.5347768 -2.6454479 
	-9.5750027 175.48743 -3.5291476 -1.6770107 -3.5278218 -1.2814322 -3.5291476 -1.6770107 
	-9.5671101 178.22159 
4.1 -9.5757761 175.11593 -3.5374949 -3.4861468 -3.5288108 -1.6893733 -3.5374949 -3.4861468 
	-9.5779299 174.0712 -3.5304701 -2.2106021 -3.5288108 -1.6893733 -3.5304701 -2.2106021 
	-9.5708708 177.67352 
5.1 -9.5772655 173.95816 -3.5401256 -4.3247667 -3.5296723 -2.0963463 -3.5401256 -4.3247667 
	-9.5794838 172.66183 -3.5316446 -2.7429773 -3.5296723 -2.0963463 -3.5316446 -2.7429773 
	-9.5742719 177.13003 
END
```

After preprocessing:
```
! AWR Design Environment (17603) Fri Mar  7 16:50:01 2025
! AWR Design Environment (17603) Fri Mar  7 16:50:01 2025
! Document = 'a_base_EM'

! nPorts: 3, nXvals: 6
VAR x1 = 20
VAR x2 = 30
BEGIN ACDATA
# GHz S DB R 50
% F N11X N11Y N12X N12Y N13X N13Y N21X N21Y N22X N22Y N23X N23Y N31X N31Y N32X N32Y N33X N33Y 
0.1 -9.5574863 179.87028 -3.5265759 -0.08911217 -3.524635 -0.042428161 -3.5265759 -0.08911217 -9.5573313 179.84258 -3.5245584 -0.05627355 -3.524635 -0.042428161 -3.5245584 -0.05627355 -9.5534535 179.93604 
1.1 -9.5640098 178.64805 -3.5296304 -0.95062039 -3.5257785 -0.45939033 -3.5296304 -0.95062039 -9.5651793 178.36214 -3.5264164 -0.60217335 -3.5257785 -0.45939033 -3.5264164 -0.60217335 -9.5582923 179.34614 
2.1 -9.5689254 177.45568 -3.5320908 -1.8007848 -3.5267782 -0.87172728 -3.5320908 -1.8007848 -9.5705455 176.91594 -3.5277627 -1.14117 -3.5267782 -0.87172728 -3.5277627 -1.14117 -9.5627602 178.77766 
3.1 -9.5730569 176.28011 -3.5347768 -2.6454479 -3.5278218 -1.2814322 -3.5347768 -2.6454479 -9.5750027 175.48743 -3.5291476 -1.6770107 -3.5278218 -1.2814322 -3.5291476 -1.6770107 -9.5671101 178.22159 
4.1 -9.5757761 175.11593 -3.5374949 -3.4861468 -3.5288108 -1.6893733 -3.5374949 -3.4861468 -9.5779299 174.0712 -3.5304701 -2.2106021 -3.5288108 -1.6893733 -3.5304701 -2.2106021 -9.5708708 177.67352 
5.1 -9.5772655 173.95816 -3.5401256 -4.3247667 -3.5296723 -2.0963463 -3.5401256 -4.3247667 -9.5794838 172.66183 -3.5316446 -2.7429773 -3.5296723 -2.0963463 -3.5316446 -2.7429773 -9.5742719 177.13003 
```

This also works with whatever file type this is (Select mdf or touchstone in MWO, file extension is s3p in this case)
```
! AWR Design Environment (17603) Fri Mar  7 16:38:23 2025
! Document = 'a_base_EM'
! nPorts: 3, nXvals: 6

VAR x1(real) = 20
VAR x2(real) = 30
BEGIN ACDATA
% freq(real) S[1,1](complex) S[1,2](complex) S[1,3](complex) S[2,1](complex) 
	S[2,2](complex) S[2,3](complex) S[3,1](complex) S[3,2](complex) 
	S[3,3](complex) 
1e+08 -0.33275499 0.00075337341 0.66630133 -0.0010362998 0.66645086 -0.00049351435 0.66630133 -0.0010362998 
	-0.33276052 0.00091428568 0.6664566 -0.00065456638 0.66645086 -0.00049351435 0.6664566 -0.00065456638 
	-0.33291016 0.00037161769 
```