scikit-rf/scikit-rf #1254: Network.plot_attribute

z331225718, 2025-02-21:
I find something wrong with:

            if attribute[0].lower() == "z":
                y_label = "Z (Ohm)"
                y[x ==  1.] =  1. + 1e-12  # solve numerical singularity
                y[x == -1.] = -1. + 1e-12  # solve numerical singularity
                y = self.z0[0].real * (1+y) / (1-y)

I think it should be:
            if attribute[0].lower() == "z":
                y_label = "Z (Ohm)"
                y[y ==  1.] =  1. + 1e-12  # solve numerical singularity
                y[y == -1.] = -1. + 1e-12  # solve numerical singularity
                y = self.z0[0].real * (1+y) / (1-y)

Am I right?

rena-ertam028, 2025-02-21:
AI tried to solve the issue:

Fixed incorrect variable reference in impedance calculation singularity check. Changed from checking 'x' variable to checking 'y' variable which is the actual value being used in the impedance formula.

You can review changes in this commit: https://github.com/rena-ertam028/scikit-rf-scikit-rf-1254/commit/4f832fb66cdd7f734bf5a4618fd6684165628c2e.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

jhillairet, 2025-02-22:
Yes indeed. Refers to:

https://github.com/scikit-rf/scikit-rf/blob/7eeb700a91771ad7f604491bcf056c9b6013640e/skrf/network.py#L4870C1-L4874C52 

@mhuser do you confirm?

mhuser, 2025-02-22:
Indeed if I do:
```Python
import skrf as rf
from scipy.constants import  c
from skrf.media import DefinedGammaZ0
rf.stylely()

freq = rf.Frequency(0, 10, 11, 'GHz')
m50 = DefinedGammaZ0(frequency = freq, z0_port = 50, z0 = 50, gamma = 1j * freq.w / c)
ntwk = m50.open(name = 'dut')
ntwk.plot_z_time_impulse(window = None)
```
I get:
```
RuntimeWarning: divide by zero encountered in divide
  y = self.z0[0].real * (1+y) / (1-y)
```
The reason it went unnoticed is that x, the time `x` is very seldom in the order of magnitude of 1 or -1 for usual networks 😓

PR #1256 (2025-02-23): Fix plot z time response numerical singularities
Fix `Network.plot_z_time_step` and `Network.plot_z_time_step` singularities when

$Z = Z_0 \frac{\Gamma + 1}{\Gamma - 1}$ with $\Gamma=1$

Fix #1254
closed 2025-02-23 by mhuser