scikit-rf/scikit-rf #1264: pdf version of the docs

bendelathouwer, 2025-03-11:
**Is your feature request related to a problem? no

**Describe the solution you'd like**
as I do most my developing on the road without internet access a pdf version of the docs would be greatly appreciated

Vinc0110, 2025-03-20:
I did some research on this and found instructions for exactly this:
https://docs.readthedocs.com/platform/stable/downloadable-documentation.html

Apparently, we need to modify our configuration of the Sphinx options to allow downloads of offline formats such as PDF or ePub:
https://docs.readthedocs.com/platform/stable/guides/enable-offline-formats.html

closed 2025-03-29 by jhillairet

jhillairet, 2025-03-29:
@bendelathouwer 

The offline format files are now available via the read the docs menu (at bottom right)

[image]

PR #1269 (2025-03-29): Adding offline formats to the documentation
- add zip html and pdf output formats
- [fix RTD config](https://github.com/astral-sh/uv/issues/10074) w/ uv to support output formats
- update the documentation (citation list)

fixes #1264