scikit-rf/scikit-rf #1266: Make Network.s_error() return a pandas Series() object instead of numpy.ndarray()

jrmagers, 2025-03-13:
**Is your feature request related to a problem? Please describe.**

When using `s_model()` to compute the error between two networks, the result is returned as a numpy array. This makes it inconvenient to plot the result vs frequency:

```Python
import matplotlib.pyplot as plt

error = ntwkA.smodel(ntwkB)

plt.plot(ntwkA.frequency.f_scaled, error)
plt.xlabel(f"Frequency {ntwkA.frequency.unit}")
```

**Describe the solution you'd like**

Instead, return the result as a pandas Series() object with the series index as the frequency. The index name can store the x-axis label. Now plotting is much easier:

```Python
error = ntwkA.smodel(ntwkB)
error.plot()
```

jhillairet, 2025-03-16:
Returning a pandas Serie or a Dataframe would change the current behaviour with all Network quantities (ex: `ntwkA.s` returns a Numpy array). In addition, there is a method called [`to_dataframe`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.to_dataframe.html) which could be envisaged as  `ntwkA.to_dataframe('s_error')` (only once the error has been calculated first... ). 

Actually I would prefer instead to have, like for all other Network quantities, directly [some plotting methods](https://scikit-rf.readthedocs.io/en/latest/tutorials/Plotting.html#Plotting-Methods), like for example: `ntwk1.plot_s_error(ntwk2)` .

Error calculation (`x_error`) and plotting methods should also be available for other network parameters such as `y`, `z`, etc.

jrmagers, 2025-03-17:
Great -- I'll create this: `ntwk1.plot_s_error(ntwk2)`.

jrmagers, 2025-03-18:
Can anyone provide some guidance on where and how to define this method?

I think it should go in as a function in `plotting.py` like this:

```Python
## specific plotting functions
...

def plot_s_error(ntwkA, ntwkB, error_function = ...):
    ...
```

But how do I get it to do operate on `self` without putting it into `network.py`?

```Python
ntwkA.plot_s_error(ntwkB, error_function = ...)
```

Thanks!

jhillairet, 2025-03-19:
I suggest to add a [plotting function ](https://github.com/scikit-rf/scikit-rf/blob/24655fad3baf54f56ea6f5ad7745f0f6f4fec247/skrf/plotting.py#L897-L901) for example `plot_error` (to be generic and allows future other network parameter error plotting) in `plotting.py`, as well as [a plotting method](https://github.com/scikit-rf/scikit-rf/blob/24655fad3baf54f56ea6f5ad7745f0f6f4fec247/skrf/network.py#L4962-L4965) (you can start with `plot_s_error` for example) in `network.py`