scikit-rf/scikit-rf #1268: setup.py missing -- no setup with pipx

mike-hb, 2025-03-29:
Thank you for reporting a bug! To help us understanding the issue and how to reproduce it, please fill the following sections as best as you can.

**Describe the bug**
git clone the actual scikit-rf
cd scikit-rf
--no setup.py available

**To Reproduce**
rm -r scikit-rf

**Expected behavior**
after installing I expect:
python3
import skrf
no error
but I got: 
Traceback (most recent call last):
  File "<python-input-0>", line 1, in <module>
    import skrf as rf
ModuleNotFoundError: No module named 'skrf'
>>>
**Data**

**System**
 - OS: opensuse Thumbleweed
 - scikit-rf -- git (cloned on 20250328)

**Additional context**
in opensuse pip install is not allowed anymore  -- a pipx installs into ~/.local/pipx or pip but not with the scikit-rf!
Thanks in advance for assisting to find out my problems

mike-hb, 2025-03-29:
Thanks for the answer -- but this does not solve the problem, that "pip install scikit-rf" is not a valid solution -- it does not work anymore : 
---------------8<-----------------
 $ pip3-3.13 install scikit-rf

[notice] A new release of pip is available: 24.3.1 -> 25.0.1
[notice] To update, run: pip3.13 install --upgrade pip
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try
    zypper install python313-xyz, where xyz is the package
    you are trying to install.
    
    If you wish to install a non-rpm packaged Python package,
    create a virtual environment using python3.13 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-rpm packaged Python application,
    it may be easiest to use `pipx install xyz`, which will manage a
    virtual environment for you. Install pipx via `zypper install python313-pipx` .

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

---------------8<-----------------
and up to now there is no rpm-package of scikit-rf ; and in addition: "make rpm" does not make it.

Any further help is very wellcome!
Michael

jhillairet, 2025-03-29:
Searching for [explanations on this error](https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3), your python environment is externally managed, in Suse, by zypper (not pip). So, to not break your system python environment, it is highly recommended to use pip in a virtual environment. (I suppose this error message is not specifically related to scikit-rf, but due to your linux distribution. Unless if you see a difference with respect to an older version of scikit-rf?)

biergaizi, 2025-09-22:
Installing packages directly via `pip` into the base system (without a `venv`) is considered a bad practice, because it may create conflicts with the system-wide package manager, thus it has been universally banned on most systems by default. It can be bypassed via `--break-system-package` but at your own risk. The most common conflict is when a system package requires dependent package, version A, but a pip package has installed version B, creating runtime problems if you're unlucky.

This is a Python / `pip` usage problem, which has nothing to do with the scikit-rf project. I think this issue can be closed.

closed 2025-09-22 by jhillairet