scikit-rf/scikit-rf #1274: IEEEP370_SE_ZC_2xThru deviates from fdf impedance in time domain

Hasko7, 2025-04-14:
The fixture generated by IEEEP370_SE_ZC_2xThru doesn’t fully follow the impedance profile of FD, especially when FD has poor causality. It looks like the method forces the step response impedance before 0 seconds to always be 50 Ohms. That causes issues in de-embedding, since the resulting fixture doesn't really match what FD represents anymore.

Here’s a small example:

```
FD = rf.Network('measurements/FD.s1p')
FD.name = 'FD'
FD_2port = rf.network.one_port_2_two_port(FD)
FD_2port.name = 'FD_2port'

se2xThru = rf.Network('measurements/ISD_THRU.s2p')
se2xThru.name = 'se2xThru'

insitu_fix = rf.Network('measurements/a1_left_dut.s2p')
insitu_fix.name = 'insitu_fix'
```
The DUT was measured in a 1-port setup using the fixture. I then turned it into a 2-port using one_port_2_two_port, so I could calculate fix1 with IEEEP370_SE_ZC_2xThru. The insitu_fix is a fix1 calculated from FD using in-situ de-embedding with impedance correction on a Rohde & Schwarz ZNA67.

Here’s what the TDR looks like in the time domain:

[image]

[image]

You can see that both FD and the in-situ fixture aren’t locked to 50 Ohms before time zero, which is pretty common in real-world measurements, since DUT structures often aren’t fully passive. But zc_fix1 just flatlines to 50 Ohms before 0 s, which seems a bit too idealized.

The de-embedding was done like this:
```
dm_zc = IEEEP370_SE_ZC_2xThru(
    dummy_2xthru=se2xThru,
    dummy_fix_dut_fix=FD_2port,
    pullback1=0,
    pullback2=0,
    leadin=1,
    NRP_enable=True,
    name='zc2xthru'
)
```

For comparison, I also calculated the fixture using a MATLAB version. It matches the calculated fixture from scikit-rf,In the time domain, the in-situ-based result nicely follows FD even before time zero. The ZC version (zc_fix1), though, just stays locked at 50 Ohms the whole time before 0 s, so it doesn’t really capture the actual behavior of FD.

[image]

[image]

Would it be possible to tweak the implementation so ZC doesn't always enforce causality or that ideal 50 Ohm behavior before 0 s? Maybe add an option to disable the causality constraint when needed?

I’ve got another measurement where the 2xThru also has weak causality like a FD (this is another one). Interestingly, in that one, nzx_fix1  follows the FD (aother one) structure before 0, while zc_fix1 again just stays at 50 Ohms. This shows that the nzc implementation differs from zc. 

[image]

[image]

I'll attach the Touchstone files here so the issue can be reproduced easily. I can not upload the touchstone files. I changed the file ending to .txt. You can rechange it to .s1p or s2p. @mhuser do you have an explanation or a solution for this?

[a1_left_dut.txt](https://github.com/user-attachments/files/19738264/a1_left_dut.txt)
[FD.txt](https://github.com/user-attachments/files/19738263/FD.txt)
[fix1_ieee370_matlab.txt](https://github.com/user-attachments/files/19738261/fix1_ieee370_matlab.txt)
[ISD_THRU.txt](https://github.com/user-attachments/files/19738262/ISD_THRU.txt)

Hasko7, 2025-04-15:
Hi @mhuser, I’ve already tried it with the parameter `leadin = 100` and even higher. The problem is that the results are not very accurate. Here are my results:

`leadin = 100`
[image]

[image]

`leadin = 200`
[image]

`leadin = 250`

[image]

`leadin = 400`

[image]

This is the FD (with `leadin= 1`):
[image]

It’s clearly visible that this does not match the behavior of FD. Especially when using even larger values for the leadin parameter. From around leadin = 250, it doesn't make sense anymore.
The issue is that the behavior of zc_fix1 does not align with that of FD. This makes the de-embedding in the frequency domain unreliable, because if the calculated fixture from ZC or NZC does not exactly match the fixture from FD, the de-embedding will not be precise.

This is what my DUT looks like after de-embedding with zc_fix1, nzc_fix1, and insitu_fix.
I did the de-embedding with:
```
dm_zc = IEEEP370_SE_ZC_2xThru(dummy_2xthru = se2xThru, dummy_fix_dut_fix = FD_2port,
                         pullback1 = 0, pullback2 = 0,
                         leadin = 1, NRP_enable = True,
                         name = 'zc2xthru')

zc_fix1 = dm_zc.s_side1
zc_fix1.name = 'zc_fix1'
zc_fix1_dc = IEEEP370.extrapolate_to_dc(zc_fix1)

dm_nzc = IEEEP370_SE_NZC_2xThru(dummy_2xthru = se2xThru,  name = 'nzc2xthru',)
ncz_fix1 = dm_nzc.s_side1
ncz_fix1.name = 'ncz_fix1'
ncz_fix1_dc = IEEEP370.extrapolate_to_dc(ncz_fix1)

zc_dut = zc_fix1.inv ** FD
zc_dut.name = 'zc_dut'
nzc_dut = ncz_fix1.inv ** FD    
nzc_dut.name = 'nzc_dut'
insitu_dut = insitu_fix.inv ** FD
insitu_dut.name = 'insitu_dut'
```

[image]

[image]

This is FD:

[image]

It is clearly visible that the DUT with the in-situ fixture shows a better match in the time domain and a better behavior in the frequency domain.

I’ve studied both algorithms, ZC and NZC, in detail and I also understand how they work. If you remember, I had discovered that in ZC, FIX 2 is incorrectly flipped.

Although ZC manages to mimic the impedance profile of FD through peeling, this imitation in the time domain only refers to the region after 0 seconds. This gives the IEEE P370 ZC a clear weakness compared to other methods like the in-situ de-embedding, which also uses impedance correction.
However, I believe this issue can be easily resolved, because we have FD or FDF, and therefore also the impedance profile of the fixture before 0 in the time domain. With this, ZC could be optimized and used more precisely for DUT de-embedding cases.
And that would also be a better solution than using the leadin parameter.

mhuser, 2025-04-19:
Hi @Hasko7, thank you for sharing these results!

> If you remember, I had discovered that in ZC, FIX 2 is incorrectly flipped

Yes, thank you so much for this!

> This` gives the IEEE P370 ZC a clear weakness compared to other methods like the in-situ de-embedding, which also uses impedance correction.

Do you know if the in-situ de-embedding algorithm is available?

I think that if the matlab version gives the same results as scikit-rf, this is at least not an implementation issue but rather a limitation of the `IEEEP370_SE_ZC_2xThru` method itself. Of course, it would be nice if there is a mean to overcome this limitation.

Indeed, the leadin does not follow the FIX-DUT impedance profile. I have added in #1276 a plot to see the extracted leadin in `IEEEP370_SE_ZC_2xThru` when `verbose = True` and exercised it on your data.

[image]

The method `peelNPointsLossless` is used to extract the portion before time 0, after the network has been shifted with a delay.
It peels the impedance iteratively but with a cascade of lossless transmission line delays of different impedances.
It is still mixed in my mind, but could it be that this algorithm cannot mimic a sloppy impedance profile?

https://github.com/scikit-rf/scikit-rf/blob/371b38ee032f7ea65d4d05bd6947c455a82c30b5/skrf/calibration/deembedding.py#L1436

Do you know what is the source of this non-causality? If I remember properly, it can be caused by limited bandwidth and discrete sampling or by measurement errors such as noise or calibration.

mhuser, 2025-10-31:
Hello @Hasko7 

> sorry for the late reply. I’ve been quite busy with my master’s thesis.

It is a good practice to attribute a high priority to such things.

> The source of the non-causality is the DUT after the fixture. When the impedance of the DUT differs from the impedance of the fixture, this non-causality occurs. You can simulate this by building an ideal transformer with the end impedance and cascading it with the fixture.

This is an interesting explanation.

Definitely, my opinion is that the leadin algorithm cannot mimic that.

I have attempted an alternative approach:

1. Isolating the return loss before time 0 using gating in the time domain
2. Deembed it from FIX_DUT_FIX
3. Invoke `IEEEP370_SE_ZC_2xThru` on the corrected FIX_DUT_FIX
4. Embed the isolated return loss in the fixture models

This gives the following results:

[image]

Are you interested in trying this?

```Python
import skrf as rf
import matplotlib.pyplot as plt
from numpy import concatenate
from numpy.fft import fftshift, ifftshift, irfft, fft
from skrf.calibration import IEEEP370, IEEEP370_SE_ZC_2xThru
rf.stylely()

FD = rf.Network('measurements/FD.s1p')
FD_2port = rf.network.one_port_2_two_port(FD)
FD_2port.name = 'FD_2port'

se2xThru = rf.Network('measurements/ISD_THRU.s2p')
se2xThru.name = 'se2xThru'

# Compute before time 0 non-causality
f = FD_2port.frequency.f
s11 = FD_2port.s[:, 0, 0]
s22 = FD_2port.s[:, 1, 1]
n = len(f)
# extrapolate to DC
dcs11 = IEEEP370.DC(s11,f)
dcs22 = IEEEP370.DC(s22,f)
# transform to time domain
t11 = fftshift(irfft(concatenate(([dcs11], s11)), axis=0), axes=0)
t22 = fftshift(irfft(concatenate(([dcs22], s22)), axis=0), axes=0)
# time gating with a rectangular window
t11[n:] = 0
t22[n:] = 0
# transform to frequency domain
sbef1 = fft(ifftshift(t11))
sbef2 = fft(ifftshift(t22))
# remove DC extrapolation
sbef1 = sbef1[1:n+1]
sbef2 = sbef2[1:n+1]
# alter perfect thrus with non-causality
bef1 = FD_2port.copy()
bef1.s[:, 0, 0] = sbef1
bef1.s[:, 1, 0] = 1
bef1.s[:, 0, 1] = 1
bef1.s[:, 1, 1] = 0
bef2 = FD_2port.copy()
bef2.s[:, 0, 0] = sbef2
bef2.s[:, 1, 0] = 1
bef2.s[:, 0, 1] = 1
bef2.s[:, 1, 1] = 0

# Deembed before time 0 non-causality
FD_2port_causal = bef1.inv ** FD_2port ** bef2.flipped().inv

# Split
dm_zc = IEEEP370_SE_ZC_2xThru(
    dummy_2xthru=se2xThru,
    dummy_fix_dut_fix=FD_2port_causal,
    leadin=0,
    NRP_enable=False,
    name='zc2xthru',
    verbose = False
)

# Embed before time 0 non-causality
dm_zc.s_side1 =  bef1 ** dm_zc.s_side1
dm_zc.s_side2 =  bef2 ** dm_zc.s_side2
dm_zc.s_side1.name = "FIX-1"
dm_zc.s_side2.name = "FIX-2"
dm_zc.plot_check_impedance(FD_2port)

# Plot
fdf = IEEEP370.extrapolate_to_dc(FD_2port)
fix1 = IEEEP370.extrapolate_to_dc(dm_zc.s_side1)
fig, ax = plt.subplots(1, 1, figsize=(10, 5))
fdf.plot_z_time_step(0, 0,
                         ax = ax, linestyle = 'dotted', color = '0.2')
fix1.plot_z_time_step(0, 0,
                      ax = ax, color = 'k')
```

Hasko7, 2025-11-23:
Hi @mhuser,
I had already tested this approach in the past and also compared it with in-situ deembedding. The in-situ method is always slightly better, because it adjusts what comes after the calculated fixture (after 0 seconds and after the fixture influence), instead of forcing it to 50 like scikit-rf.
In this case, the fixture creates a kind of slope that corresponds to the fixture + DUT (FD).

[image]

[image]

[image]

Here in the frequency domain, you can also see that the in-situ result is noticeably more linear (without the ringing), which better matches the measured DUT (in this case an ASIC chip).

[image]

I also tested this on real measurements with a 2-port DUT. What I observed is that after calculating Fixture 1 and Fixture 2, the impedance is not forced to 50 Ohm, nor is there a slope like in this example. Instead, the fixture with impedance correction follows the impedance from the midpoint to the endpoint of the FDF, so that at the last impedance point, Fix1 and the FDF converge.
In this case as well, the in-situ method performed better, even when taking the impedance points from the FDF before 0 s. This shows that there is clearly more behind the in-situ deembedding with impedance correction.

Unfortunately, I no longer have the measurement data, but it would definitely be worth taking a closer look at this. I will try to retrieve those measurement results again. Unfortunately, I am currently very busy and have little time for it.