scikit-rf/scikit-rf #1281: After Embedding two s4p files, write_touchstone() not creating file properly

dshapoval1988, 2025-05-04:
Hello, after using the embedding ** operand on two networks created from s4p files, the action of write_touchstone to a new s4p file generates file that does not show the real behavior of the embedded transition line. Crossed the data with PLTS embedding output.
Can you please assist?

jhillairet, 2025-05-04:
Can you please give a minimal example?

The following works for me:

```python
import skrf as rf
import numpy as np

# create two 4-port dummy networks
rng = np.random.default_rng()
freq = rf.Frequency(1, 10, npoints=10, unit='GHz')
ntwk1 = rf.Network(frequency=freq, s=rng.random((10, 4, 4)))
ntwk2 = rf.Network(frequency=freq, s=rng.random((10, 4, 4)))

# cascade and write
ntwk = ntwk1 ** ntwk2
ntwk.write_touchstone('test.s4p')

# test
_ntwk = rf.Network('test.s4p')
assert ntwk == _ntwk  # OK
```

dshapoval1988, 2025-05-05:
Hello so if I use the two s4p files and perform embedding on them with plts I get reasonable S parameters. See picture **Embedding_by_PLTS**
However when I perform the following script the I don't get a proper output as PLTS. See picture **Embedding_by_Python** 

import skrf as rf

network1 = rf.Network('file1.s4p')
network2 = rf.Network(file2.s4p')

embedded_network = network1 ** network2
embedded_network.write_touchstone('Embedded_by_python.s4p')

[image]
[image]

jhillairet, 2025-05-05:
Does your 4ports network follow the expected port numbering, cf. https://scikit-rf.readthedocs.io/en/latest/tutorials/Connecting_Networks.html#Cascacing-2N-port-Networks ?

dshapoval1988, 2025-05-05:
How can I check this ? 
My files are made by measuring s4p files by Keysight PNA

mhuser, 2025-05-05:
Hello, please mind that `**` or [`cascade`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.cascade.html) need a specific port numbering convention for 2-N ports. Maybe in your case you need to [`renumber`](https://scikit-rf.readthedocs.io/en/latest/api/generated/skrf.network.Network.renumber.html#skrf.network.Network.renumber) first ?
```Python
"""
0  1  to 0  2
2  3      1  3
"""
network1 = rf.Network('file1.s4p').renumber([0, 1, 2, 3], [0, 2, 1, 3])
network2 = rf.Network(file2.s4p').renumber([0, 1, 2, 3], [0, 2, 1, 3])

[...]
# back
embedded_network.renumber([0, 2, 1, 3], [0, 1, 2, 3])

```

> How can I check this ?

Usually you need to know what port of the VNA is what port of the s-parameter file and what is your wiring, maybe by default port1 is 1 and so on.

dshapoval1988, 2025-05-05:
The VNA measurement set up as PORT1=>PORT2, PORT3=>PORT4 for both files.
It looks like the renumber is helping however there is still difference between the python embedding and PLTS. 

[image]

dshapoval1988, 2025-05-05:
Looks like the renumber worked fine. The delta was due to PLTS embedding was not done right.

jhillairet, 2025-05-06:
Great to hear this. Can you close the issue?

closed 2025-05-07 by dshapoval1988