scikit-rf/scikit-rf #1294: Comment attribute does not read comments

patfhr, 2025-07-11:
**Describe the bug**
I have .s2p files with a bunch of metadata in the comments. They usually have a format like:
! VNA-Name
!Temperature: 21 °C

However, when reading this files the comments attribute does not contain any data:
sparam = rf.Network(str(path))
print(sparam.comments)

**To Reproduce**
Load an .snp file with comments and read the comments attribute.

**Expected behavior**
The comments attribute should show all comments in list format.

**System**
 - OS: Windows 10
 - scikit-rf version 1.2.0

jhillairet, 2025-07-12:
Indeed, I can reproduce it. Seems only the first line of comment is read.

jhillairet, 2025-07-12:
> **Describe the bug** I have .s2p files with a bunch of metadata in the comments. They usually have a format like:

Are the comments located in between data?

I can read comments from Touchstone files. Actually, the  `.comments` parameters is a concatenated string of all comments located before the data.

Can you try with the latest scikit-rf version?

FranzForstmayr, 2025-07-14:
your example actually shows comments after the option line (# Hz S ....).

You can access your comments with `network.comments_after_option_line`.

```python
import skrf as rf

net = rf.Network("test.s2p")
print(net.comments_after_option_line.split("\n")
```

Usually skrf expects comments before the option line, in this case the comments are preserved when writing to a file again.

patfhr, 2025-07-14:
Thanks, that's working :)