scikit-rf/scikit-rf #1322: Confusing and Inconsistent Variable Naming nports vs n_ports

biergaizi, 2025-09-28:
In scikit-rf, sometimes the number of ports is specified using the variable name `n_ports`, sometimes it's specified as `nports`. For example, `media.open()` accepts a `nports` while `media.random()` accepts a `n_ports`. This is inconsistent and confusing.

The usage `n_ports` should be completely deprecated in the codebase.

Vinc0110, 2026-02-18:
That's ok for me. Consistency is important, so let's have 'nports' everywhere.

biergaizi, 2026-03-02:
> That's ok for me. Consistency is important, so let's have 'nports' everywhere.

I'd like to add that I didn't proposing changing `n_ports` to `nports` without a reason. If you grep the keyword `n_ports` in the codebase, you would see that `random()` is the only public API that uses the argument `n_ports`, all other APIs have `nports`  (although both `n_ports` and `nports` are used as internal variable names). Do we break 1 API, or do we break all APIs, the choice is obvious. 

> This looks like a fun and easy first PR for the project.

While the problem is easy, I won't say it's "fun", so this is probably why nobody has submitted a patch. To do it properly, the PR should not just change the variable name, but instead.

During the transition stage:

1. One cannot just change `n_ports` to `nports`, the old `n_ports` must also be redirected to `nports`. Leaving both `nports` and `n_ports` in the argument list would be confusing and sensitive to argument position, so it's better to be handled as a special case via `**kwargs`.

2. If `n_ports` is used, the program should raise a clearly-worded runtime deprecation warning. The same deprecation warning shoud appear in the docstring of `random()`.

3. All uses of `n_ports` in documentation, examples, and test cases must be updated.

During the removal stage:

1. If `n_ports` is used, the program should raise a hard error instead of a warning. This is actually not easy because scikit-rf allows one to pass unused `kwargs` around without raising any error, so it may be special-cased.

If you're willing to make a contribution that fits the criteria, patch welcome.